timvideos/HDMI2USB-litex-firmware #211: Capture is over bright / over saturated

mithro, 2016-02-03:
Capture from many laptops is over bright / over saturated.

The root cause seems to be the `Byte 23` - "Display gamma, datavalue = (gamma*100)-100 (range 1.00–3.54)" (see https://en.wikipedia.org/wiki/Extended_Display_Identification_Data).

mithro, 2016-02-03:
@xfxf tried to fix this problem at https://github.com/timvideos/HDMI2USB-misoc-firmware/commit/d2cbd1dd14d42fb786c8798df7deff7bbbe4eebe and https://github.com/timvideos/HDMI2USB-misoc-firmware/pull/205

mithro, 2016-02-03:
This fix seems to crash gnome-shell somehow?

joeladdison, 2017-01-09:
I found the documentation for the Generic RGB Profile that we normally use for Macs at conferences - https://developer.apple.com/library/content/qa/qa1430/_index.html

It specifies that the gamma is 1.8, which is 0x50 in EDID terms.

mithro, 2017-03-19:
@xfxf  & @joeladdison - I believe this is fixed now? Reopen it if it isn't...

closed 2017-03-19 by mithro

joeladdison, 2017-03-20:
@mithro & @xfxf Yes, I think this has been fixed now. I changed the gamma value and it seems to have improved the capture brightness.

xfxf, 2017-03-20:
I've not noticed any problems, before it was super obvious, so I believe fixed, yep.