timvideos/HDMI2USB-litex-firmware #216: Make the HDMI2USB work when using a or1k soft-cpu rather than the lm32 version

mithro, 2016-03-09:
MiSoC current supports two soft CPUs (https://m-labs.hk/gateware.html);
> CPU options:
> * LatticeMico32, modified to include an optional MMU (experimental).
> * mor1kx, a better OpenRISC implementation.

The reasons for us currently choose the lm32 soft-cpu over the or1k soft-cpu where;
 * lm32 doesn't require patching gcc
 * lm32 is slightly smaller in resource requirements than the or1k
 * lm32 is the default option in MiSoC :)
 
However, the or1k seems like it has more people developing for it compared to the lm32. There seem to be more advanced Linux ports for example. Having or1k support would enable us to try and capture some of that effort.

mithro, 2016-03-09:
FYI - @shenki

mithro, 2016-04-03:
WIP stuff here -> https://github.com/mithro/HDMI2USB-misoc-firmware/tree/or1k

binutils and gcc for or1k have been added (and are the same version as lm32 versions) to conda https://github.com/timvideos/conda-hdmi2usb-packages

Added the base target to build for or1k - see example at https://travis-ci.org/mithro/HDMI2USB-misoc-firmware/builds/120395735

gateware and firmware compile for the base target but have not been tested at all.

enjoy-digital, 2016-04-03:
That seems fine, on my side I'm going to continue working on #229 next week. With these two points solved we will have a good basis for GSoC and to continue work on the firmware.

mithro, 2016-04-17:
My WIP branch is slowly getting there. Hopefully with #257 we will be able to figure out why the firmware is locking up.

mithro, 2017-10-09:
This is working now with the "Exception Stack Size Fix":
 * LiteX - https://github.com/enjoy-digital/litex/pull/31 - https://github.com/enjoy-digital/litex/commit/b1b6a741702a7ca1bf5b021f7f8504784a26aa19
 * misoc - https://github.com/m-labs/misoc/pull/56 - https://github.com/m-labs/misoc/pull/56/commits/98dd745548319242399e8ff1e7137dbb73e3cb21

Yay!

mithro, 2017-10-09:
Fixed!

closed 2017-10-09 by mithro