timvideos/HDMI2USB-litex-firmware #218: hdmi2usb.tv/firmware does not tell me how to update the firmware

CarlFK, 2016-03-18:
https://hdmi2usb.tv/firmware/

I want to upgrade the firmware in my Opsis board.  I have done it before, but only when I accidentally got it into the mode needed, and I have no idea how to get it into that mode.  I think it is dependant on what is plugged in when I power it on.

I hear fxload is involved.

CarlFK, 2016-03-18:
HDMI2USB>help
...
debug fx2_reboot firmware      - reboot the FX2 USB IC into firmware            

HDMI2USB>debug fx2_reboot firmware                                              
Possible FX2 Firmware:                                                          
 [ ] USB JTAG (13C2)                                                            
 [*] HDMI2USB (1BAC)                                                            

I am not sure how to make this work.

mithro, 2016-03-20:
BTW For everyone else, the code can be found at https://github.com/timvideos/HDMI2USB-misoc-firmware/blob/master/firmware/lm32/ci.c#L660

You shouldn't be able to see any messages after the "fx2: Turning off" as you are using the FX2 to see those messages. It sounds like it was not successful in rebooting the FX2. Did it eventually time out?

mithro, 2016-03-20:
BTW The `debug fx2_reboot` command hasn't been the most reliable, it might take a couple of attempts at the moment.

CarlFK, 2016-03-22:
Still not sure what to do.
```
DMI2USB>debug fx2_reboot usbjtag                                                
fx2: Turning off.                                                               
fx2: Turning on.                                                                
fx2: Waiting for FX2 to load firmware.                                          
```
While that is happening:
```
carl@twist:~/temp/HDMI2USB-firmware-prebuilt/archive/master/v0.0.0-604-ge452948/opsis/hdmi2usb$ openocd -f board/digilent_atlys.cfg -c "init; pld load 0 opsis_hdmi2usb-hdmi2usbsoc-opsis.bitt; exit"
Open On-Chip Debugger 0.9.0+git20151228 (2016-01-26-04:52)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Adapter driver 'usb_blaster' did not declare which transports it allows; assuming legacy JTAG-only
Info : only one transport option; autoselect 'jtag'
Warn : incomplete ublast_vid_pid configuration
jtagspi_program
Info : usb blaster interface using libftdi
Error: unable to open ftdi device: device not found
```
I am guessing I should see a different usb id?
I don't. this never changes: 
```
watch -d lsusb
...
Bus 001 Device 012: ID 2a19:5442
```
Also, if I power cycle I only see that ID go away and come back, I never see any other ID.

mithro, 2016-03-23:
You shouldn't see anything past `fx2: Turning off.` As I mentioned you are using the FX2 to view this output and it is the FX2 which is being reset here.

Can you check the jumpers on the board and make sure that the Cypress Reset line is properly connected?

olasd, 2016-04-28:
I seem to be hitting the same fx2_reboot issue; My JFX2-RST jumper is in the 2-3 position. Full output : 

```
HDMI2USB>debug fx2_reboot usbjtag
fx2: Turning off.
fx2: Turning on.
fx2: Waiting for FX2 to load firmware.
fx2: Waiting at 00 (end: 13C2)
fx2: Waiting at 00 (end: 13C2)
fx2: Waiting at 00 (end: 13C2)
fx2: Waiting at 00 (end: 13C2)
fx2: Waiting at 00 (end: 13C2)
fx2: Waiting at 00 (end: 13C2)
fx2: Waiting at 00 (end: 13C2)
fx2: Waiting at 00 (end: 13C2)
fx2: Waiting at 00 (end: 13C2)
fx2: Waiting at 00 (end: 13C2)
fx2: Timeout loading!
```

mithro, 2016-04-29:
What firmware version are you using?

olasd, 2016-04-29:
```
HDMI2USB>version
gateware revision: 0607e926
firmware revision: 0607e926, built Jan  8 2016 23:55:47
```

mithro, 2016-04-29:
You'll need a newer version.