timvideos/HDMI2USB-litex-firmware #219: serial port drops characters

CarlFK, 2016-03-22:
When I send characters too fast, most of them get dropped.

I think the problem was described as: there is no flow control.

Seems like there have been more than one work around.

Can this be fixed on the device side?

mithro, 2016-03-23:
I assume you mean on the Opsis board?

mithro, 2016-03-23:
This will be improved (but not totally eliminated) by the FX2 code refactor at https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/149

We can't implement hardware flow control as the pins needed are not connected. Software flow control wouldn't fix the problem you are encountering either, as cat wouldn't use it.

mithro, 2017-05-19:
This issue was moved to timvideos/HDMI2USB-fx2-firmware#13

closed 2017-05-19 by mithro