timvideos/HDMI2USB-litex-firmware #224: Make the HDMI2Eth target full functional

mithro, 2016-03-24:
The HDMI2Eth target doesn't really work yet, to make it work it would need to do the following;
 * Stream a RTP of the video stream out the Ethernet (see start at https://github.com/timvideos/HDMI2USB-misoc-firmware/blob/master/gateware/streamer/rtp.py)
   * Would be good if you could stream both compress and uncompress video.
   * Would be good if you could stream more then one input source (assuming the bandwidth allows).
 * Allow control of the board via telnet, HTTP or other network protocol.
 * Use DHCP to get an IP address.
 * Support both the Atlys and Opsis board (and maybe others). See https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/223
 * Supports operating at Gigabit speeds
 * (Optional) Allows streaming an RTP stream *into* the board as another source that can be output.

Useful resources might be;
 * @tariq786 's project to get Ethernet working in the previous firmware.
   * http://hdmi2ethernet.blogspot.com.au/
   * https://docs.google.com/document/d/1RfgA-Fi7JEO8Z5PPYnO2wFwvrL0wVjoxkvNZ8yqMA-Y/edit
   * https://github.com/tariq786/HDMI2USB
   * https://docs.google.com/document/d/1oknHWujhhMfLyjyc4yfocAlLUUg15ptux0R3oOo5Bms/edit
 * M-Lab's artiq - https://github.com/m-labs/artiq
   * They use lwip - https://github.com/m-labs/artiq/search?utf8=%E2%9C%93&q=lwip
   * Some discussion at http://irclog.whitequark.org/m-labs/2015-10-26#14423384;

It might be good for the firmware running on the soft-CPU to be running a full Linux (https://github.com/timvideos/getting-started/issues/30) or RTEMs OS so that their TCP/IP stack can be reused.

mithro, 2016-03-24:
There are also other options, @sbourdeauducq suggested taking a look at https://github.com/tass-belgium/picotcp

danielkucera, 2019-06-05:
Is this thing still alive?

mithro, 2019-06-05:
@danielkucera It is still something we very much want to do but have not had time to finish.

tariq786, 2019-06-05:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/timvideos/HDMI2USB-litex-firmware/issues/224?email_source=notifications\u0026email_token=AAAEXI2EEQGQUBCNYC36DDLPY7PSNA5CNFSM4B664P62YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODXAGDPY#issuecomment-499147199",
> "url": "https://github.com/timvideos/HDMI2USB-litex-firmware/issues/224?email_source=notifications\u0026email_token=AAAEXI2EEQGQUBCNYC36DDLPY7PSNA5CNFSM4B664P62YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODXAGDPY#issuecomment-499147199",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

This is normal when you want to do too many things in parallel ;) Please excuse any typos.