timvideos/HDMI2USB-litex-firmware #246: or1k does not fit on minispartan6+

shenki, 2016-04-10:
```
ERROR:Place:836 - Not enough free sites available for the components of the
   following type(s).
      LUTM   	Number of Components 10816   	Number of Sites 7328

ERROR:Place:375 - The design does not fit in device.
    Total LUT Utilization      : 20004 out of 30064
    LUTs used as Logic         : 9188
    LUTs used as Memory        : 10816
    FF Utilization             : 2579 out of 30064
```

enjoy-digital, 2016-04-10:
Can you share your design so that I can have a look? Also are you using the LX9 or LX25 fpga?

mithro, 2016-04-10:
Is this with embedded firmware? What else is in the SoC you are using?

Not sure there is a lot we can do about this?

mithro, 2016-04-10:
I'm pretty sure that @shenki is using an LX25?

enjoy-digital, 2016-04-10:
Also if you want to work on software, we will probably have to use the flash to store software instead of the embedded ram.

mithro, 2016-04-10:
If you are working on software you probably don't want embedded firmware or spi flash. Instead load via the serial port every time.

shenki, 2016-04-10:
Here is my design: https://github.com/shenki/HDMI2USB-misoc-firmware/tree/ms6

mithro, 2016-04-11:
I confirmed with @shenki that the issue here is trying to embed the firmware into the gateware. The minispartan6+ doesn't have enough block ram hence it seems to tries and implement it using the LUTMs, which it also doesn't have enough.

When you remove the embedded firmware, it boots to the BIOS but @shenki wasn't able to get the firmware to boot because it currently assumes that it was embedded in the gateware.

shenki, 2016-04-11:
Right. Once I removed the frimware from the gateware build I could produce a bitstream that worked, and the BIOS worked, but loading the firmware over serial failed.

This was because our firmware linker script `firmware/softcpu/linker.ld` places all the text in a region called `firmware_ram`. This region is only defined in the case when we are building the firmware into the bitstream. Linking warns but does not fail, so we end up with the firmware linked at 0x00000000 instead at the expected SDRAM location.

The solution is to emit the correct MEMORY section for the linker script that describes where our RAM is even when we're not embedding the firmware in the gateware.

mithro, 2016-04-12:
This is related to https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/236

mithro, 2016-04-12:
```
(H2U B=minispartan6 T=base) tansell@tansell-z620-l2:~/foss/timvideos/hdmi2usb/HDMI2USB-misoc-firmware-ms6$ make connect-lm32                 third_party/misoc/tools/flterm --port /dev/ttyUSB1 --speed 115200
[FLTERM] Starting...

MiSoC BIOS   http://m-labs.hk
(c) Copyright 2007-2014 Sebastien Bourdeauducq
Revision 4b008017 built Apr 12 2016 23:01:13

BIOS CRC passed (e52899ab)
Running on MiSoC rev. 4b008017 (sysid:UN) at 80MHz
Initializing SDRAM...
Memtest OK
Automatic boot in 2 seconds...
Q/ESC: abort boot
F7:    boot from serial
Booting from flash (30180000)...
Loading 18900 bytes from flash...
Executing booted program.

HDMI2USB firmware  http://timvideos.us/
hardware version info
===============================================
           DNA: 013ff280bf64b27e

gateware version info
===============================================
      platform: minispar
        target: BaseSoC
      revision: e4778f01ff677abf435b8764cb12999e39a84446
misoc revision: 4b008017

firmware version info
===============================================
    git commit: b01a664e6d34391670135fb1e7c1ede695febcc2
    git branch: shenki-minispartan6+
  git describe: v0.0.0-686-gb01a664-dirty
```

mithro, 2016-04-16:
This is now supported. When we merge the minispartan6+ support, it should all work.

closed 2016-04-16 by mithro