timvideos/HDMI2USB-litex-firmware #281: Having the serial port open prevents video from streaming

xfxf, 2016-08-07:
Steps to repo;

 1) Connect the HDMI2USB
 2) Connect to the serial port using flterm
 3) Try and get video from the video device

On newer Ubuntu versions you are the following output while trying to get video;
```
[19325.775956] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
[19330.773331] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
[19335.770452] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
[19340.767697] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
```
If you then kill flterm, you can then get video from the video device if you wait long enough. The kernel output will look like this;
```
[19424.301449] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
[19429.298562] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
[19434.295781] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
[19443.310507] usb 2-1.1: reset high-speed USB device number 24 using ehci-pci
```
After the reset line, the video will start streaming in.

From the point you have gotten video you can have both the serial and video running. Stopping / starting the programs (the video or serial) has no effect.

It works first time if you do;

 1) Connect the HDMI2USB
 2) Try and get video from the video device
 3) Connect to the serial port.

xfxf, 2016-08-07:
The gstreamer pipeline dies with the following error;
```
obtained NetClientClock from host <NetClientClock object at 0x7f80747bd240 (GstNetClientClock at 0x1d98290)>
waiting for NetClientClock to sync…
starting pipeline
playing
Received Error-Signal
Error-Details: #13: gstv4l2object.c(3274): gst_v4l2_object_set_format_full (): /GstPipeline:pipeline0/GstV4l2Src:videosrc:
Call to S_FMT failed for MJPG @ 1280x720: Input/output error
```

xfxf, 2016-08-07:
1) Connect the HDMI2USB
2) Connect to the serial port using flterm, close flterm
3) Try and get video from the video device
4) See -110 errors
5) See the reset
6) Video starts working

-----
1) Connect the HDMI2USB
2) Start the video stream, stop video stream.
2) Connect to the serial port using flterm, close flterm
3) Try and get video from the video device
4) See -110 errors
5) See the reset
6) Video starts streaming again.

mithro, 2016-08-07:
It is likely the #148 issue is related here. 

It is also likely this will be fixed by #149.

mithro, 2017-05-19:
This issue was moved to timvideos/HDMI2USB-fx2-firmware#14

closed 2017-05-19 by mithro