timvideos/HDMI2USB-litex-firmware #303: snow bars

CarlFK, 2016-11-17:
wonky bytes are showing up in the outputs, both hdmi and encoder.  it seems to be random bytes scattered around, often so little it is hard to even notice.

It happens when the pattern is the source.

it seems to get very worse when a 2nd hdmi input is connected. 

I am not aware of testing if it goes away when there are no inputs.

mithro, 2016-11-17:
@CarlFK Can you dump some pictures into this bug report?

mithro, 2016-11-29:
The amount of "snow" seems to be proportional to the amount of traffic on the memory bus. The more inputs/outputs enabled - the more the snow appears.

mithro, 2016-11-29:
We believe this is caused by an issue in the DDR memory controller.

We are currently hoping that when the rebase onto the latest litex with the latest litedram will fix this problem (the work for which is happening here -> https://github.com/enjoy-digital/opsis-soc).

mithro, 2017-03-16:
With the merge of the litex firmware, this has been fixed now!

closed 2017-03-16 by mithro