timvideos/HDMI2USB-litex-firmware #304: Unrealistically high refresh rate

stefanor, 2016-11-18:
```
input0:  1280x720 (@ 74444 kHz)
input1:  0x0 (@ 0 kHz)
output0: 1280x720@60Hz from input0
output1: 1280x720@60Hz from pattern
encoder: 1280x720 @ 0fps (0Mbps) from input0 (q: 50)
ddr: read: 1760Mbps  write:  884Mbps  all: 2644Mbps
```

I don't believe it's actually getting 74MHz video

mithro, 2016-11-19:
What is the debug output showing? I'm guessing a very high WER rate....

mithro, 2016-11-29:
Oh, that isn't the refresh rate (which is why it is seperate and in brackets). It is the pixel clock. 74MHz is about right for 720p video.

You should get a pixel clock value even when the video is out of spec (iE 1080p60) or experiencing high WER (even when the resolution values are nonsense). If this value is not stable or showing up, then something is seriously wrong with the signal.