timvideos/HDMI2USB-litex-firmware #305: An opsis gets lots of noise

stefanor, 2016-11-19:
One of our opsis won't run firmwares newer than `0.0.0-654-gbca243b`, and on that version, we get quite a bit of noise, when two HDMI inputs are connected.

WER is 0.

See attached video (in a ZIP, because GitHub...)
[two-hdmi.zip](https://github.com/timvideos/HDMI2USB-misoc-firmware/files/601537/two-hdmi.zip)

mithro, 2016-11-29:
There appears to be two issues here;
 1. The snow on the video output - I believe that is the same as https://github.com/timvideos/HDMI2USB-misoc-firmware/issues/303 - but won't be able to prove that until the merge happens and we confirm that problem is fixed for most people.
 2. The Opsis won't run newer firmwares.

I'm going to make this bug just refer to option 2.

mithro, 2016-11-29:
@stefanor Can you put some more information here about your debugging regarding the Opsis not running newer firmware?

stefanor, 2016-11-30:
Flashed it with `--flash-gateware` and it wouldn't boot afterwards. Had to put the FX2 into recovery mode to re-flash. Didn't try `--load-gateway`, probably should have played with it a little more...

mithro, 2017-03-19:
@stefanor Can you get this checked with the current firmware?

mithro, 2017-08-04:
@stefanor Ping?

stefanor, 2017-08-04:
Can't reproduce with the litex firmware.

closed 2017-08-04 by stefanor

mithro, 2017-08-04:
I wonder if it was something to do with the jumper?