timvideos/HDMI2USB-litex-firmware #306: First-time build process is very convoluted, some build scripts don't behave

jaseg, 2016-12-07:
This is due to a number of issues:

* Dependency on legacy ```misoc```
* Use of ```curl|bash``` anti-pattern (serveral places)
* Large bunch of things in ```third_party```, and no checking of submodules in place
* Mixing dependencies on system-wide ```misoc``` and ```third_party/misoc```
* Lack of documentation on *actual* build prereqs, and ```setup-env.sh``` is missing some
* Overly narrow version checks in ```setup-env.sh```
* Ubuntu-specific dependencies (```ppa```, ```apt-get```, ```tempfile```)
* Use of third-party package manager (```conda```)
* Scripts calling sudoing into stuff, even *adding a PPA repo*

Thank you very much for this awesome project. I think this is a great project and I think you are providing a very valuable contribution to the community. But that last point (```sudo add-apt-repository -y ppa:...```) is beyond WTF-is-this-doing-to-my-poor-linux-level to me. I'll submit a PR when I have some spare time, probably some time in January.

mithro, 2016-12-07:
Hi @jaseg,

I'm working on improving the setup instructions and scripts but have yet to finish them (I have been concentrating on moving to the new ```misoc```/```litex```). You can find the updated instructions at https://github.com/mithro/HDMI2USB-misoc-firmware/tree/docs-scripts-update/scripts

There are a couple of restrictions however;
 * We are going to continue to use conda to provide the build environment.
 * We will continue to require exact revisions of all core dependencies. 

> This is due to a number of issues:
> 
> * Dependency on legacy ```misoc```

We are working on this however it has been a major effort, this will hopefully finally be fixed in the next couple of weeks.

*However*, we will never support using something other than the version provided in third_party. This is good practice for projects which have complicated interactions between components.

> * Use of ```curl|bash``` anti-pattern (serveral places)

People are hopeless at following instructions, computers are good at doing so. This is the reason everyone uses the ```curl|bash``` pattern.

> * Large bunch of things in ```third_party```, and no checking of submodules in place

I have no idea what you are saying here?

> * Mixing dependencies on system-wide ```misoc``` and ```third_party/misoc```

We don't depend on system-wide ```misoc```. We **only** use ```third_party/misoc```.

(In fact we don't use the system installed python *at all*, we only use the conda provided Python.)

> * Lack of documentation on *actual* build prereqs, and ```setup-env.sh``` is missing some

```setup-env.sh``` is not missing anything for building the firmware. It is used by the CI system and developers regularly

> * Overly narrow version checks in ```setup-env.sh```

This is important for reliably reproducing working firmware. We have lost a large number of developer hours tracking down problems caused by people using different GCC versions (or requiring a newer version and people not having it installed).

> * Ubuntu-specific dependencies (```ppa```, ```apt-get```, ```tempfile```)

We currently only support developing on Ubuntu. Nobody else has bothered to contribute support for other Linux versions. I would welcome patches.

> * Use of third-party package manager (```conda```)

Nobody has found a better solution. conda is pretty agnostic to the Linux distro which is more portable than using deb packages.

Below you says "don't add a PPA or apt-get install things" that would be the alternative option to conda.

> * Scripts calling sudoing into stuff, even *adding a PPA repo*

All the sudo stuff is in a separate script so you can examine it.

> Thank you very much for this awesome project. I think this is a great project and I think you are providing a very valuable contribution to the community. But that last point (```sudo [...]

I welcome your contributions. Please based them off the doc-script-update branch I linked above.

mithro, 2016-12-08:
@jaseg - You should also come join us on IRC in the [#timvideos on irc.freenode.net](webchat.freenode.net/?channels=#timvideos)

mithro, 2017-03-30:
Hi @jaseg - The latest version of the scripts have been merged.

closed 2017-03-30 by mithro