timvideos/HDMI2USB-litex-firmware #307: Weird tiling effect when receiving a 640x480 signal while set to 720p mode

xfxf, 2017-01-12:
Mode on the Opsis was set to 720p. Computer was forced to output 640x480. Get tiled output!?

@xfxf will attach some videos/photos.

xfxf, 2017-01-12:
[image]

https://goo.gl/photos/wReAzoa1hoz6eXRw9

enjoy-digital, 2017-01-12:
That's what we can expect of what is implemented. (You have in fact the two 640pixels input lines in the 1280pixels output line). What behaviour are you expecting? Should be change that? Or should we consider that your test case is something that is not supposed to be supported (the video source is not outputing what it should).