timvideos/HDMI2USB-litex-firmware #322: Fix the pre-built download script

mithro, 2017-04-04:
Should be able to just download the correct directory from pre-built using svn export

ishan98, 2017-05-03:
Are you talking about the mismatch in the directory during the time of the execution of the **download-env.sh** file in the installation process within the following line of code.
```
for LITE in $LITE_REPOS; do
	LITE_DIR=$THIRD_DIR/$LITE
	(
		cd $LITE_DIR
		python setup.py develop
	)
	check_import $LITE
done
```

mithro, 2017-05-04:
No.

I'm talking about this script [`./scripts/download-prebuilt.sh`](https://github.com/timvideos/HDMI2USB-litex-firmware/blob/master/scripts/download-prebuilt.sh). 

The idea behind this script is that it downloads prebuilt gateware from the https://github.com/timvideos/HDMI2USB-firmware-prebuilt repository so that you can do firmware development without needing the Xilinx tools to compile the gateware.

With the change in how we manage the ./build/ directory, we should be able to use something like in the ["download" code in `build-litex-qemu.sh` script the qemu-litex repo](https://github.com/timvideos/qemu-litex/blob/master/build-litex-qemu.sh#L12-L53).

PR #340 (2017-08-11): Fix the download-prebuilt script.
In theory, it will download the latest git revision in your branch which
is upstream.

Fixes #322.
commit f212ba81 (2017-08-11): Fix the download-prebuilt script.

In theory, it will download the latest git revision in your branch which
is upstream.

Fixes #322.
closed 2017-08-11 by mithro