timvideos/HDMI2USB-litex-firmware #346: enabling a second output makes output unstable

stefanor, 2017-08-10:
At DebConf17 we were getting a lot of HDMI resets (or something) on one of our outputs, when both were enabled (+ USB streaming)

This lead to 9eb83b4a6645950a13e466893ff20b9683c4ed15 which resolved the problem.

mithro, 2017-08-13:
You were getting unflows when this was happening, right?

Can you get me the debugging output from when it was occurring?

stefanor, 2017-09-17:
@mithro took some slow-motion videos of this: 
* https://photos.app.goo.gl/H7XWpXDICjwVe8Z22

We could reproduce it at 720p50 with only 1 enabled output. 2 outputs + streaming makes it happen very often.

stefanor, 2017-09-18:
> You were getting underflows when this was happening, right?

Looks like it.
```
H2U 00:02:08>s
input0:  0x0 (@48.86 MHz) (capturing)
input1:  1280x720 (@74.24 MHz) (capturing)
output0: 1280x720@60.00Hz from input1 (underflows: 0)
output1: 1280x720@60.00Hz from input1 (underflows: 125)
EDID primary mode:   1280x720@60.00Hz
EDID secondary mode: 1280x720@60.00Hz
encoder: 1280x720 @ 24fps from input1 (q: 85)
ddr: 
```

```
hardware version info
===============================================
           DNA: 01325c9b471dfc80
           MAC: d8:80:39:ec:15:19

gateware version info
===============================================
      platform: opsis
        target: HDMI2USB
      revision: 4242bb5a89f00437f1580e31e8f762cbcab100d2

firmware version info
===============================================
      platform: opsis
        target: hdmi2usb
    git commit: 4242bb5a89f00437f1580e31e8f762cbcab100d2
    git branch: master
  git describe: v0.0.3-745-g4242bb5-dirty
    git status:
    --
     M ../../../../third_party/litepcie
     M ../../../../third_party/litesata
     M ../../../../third_party/liteusb
    --

         built: Aug 11 2017 04:51:06
        uptime: 00:02:31
-----------------------------------------------
```