timvideos/HDMI2USB-litex-firmware #347: USB streaming is corrupted on Atlys

stefanor, 2017-08-10:
With firmware build v0.0.3-730-gf9c031d and FX2 firmware (`hdmi2usb.hex`) from v0.0.2-81-gd0d3aea (as well as a locally built one from HEAD) we get this over the USB stream.

[image]

The source is the Debian [stretch background](
https://wiki.debian.org/DebianArt/Themes/softWaves)

stefanor, 2017-09-18:
Reproduced with `v0.0.3-745-g4242bb5` and a hdmi2usb.hex built from timvideos/HDMI2USB-fx2-firmware@8359b974242ec81fe93627053f70e4d10c2d8844 with `BOARD=atlys`

stefanor, 2017-10-29:
It appears to be broken in the first 0.0.3 build (v0.0.3-664-gfab22b9), again with `hdmi2usb.hex` from v0.0.2-81-gd0d3aea. Tested by generating the `firmware.bin` from `firmware.fbi` `dd if=firmware.fbi of=firmware.bin bs=1 skip=8` and loading this with `flterm`.

stefanor, 2017-10-29:
```
hardware version info
===============================================
           DNA: 0137ce976e3a3dec
           MAC: Unknown

gateware version info
===============================================
      platform: atlys
        target: HDMI2USB
      revision: fab22b9b3252d5ea8b7cf1cfd1d14d86929fd82e

firmware version info
===============================================
      platform: atlys
        target: hdmi2usb
    git commit: fab22b9b3252d5ea8b7cf1cfd1d14d86929fd82e
    git branch: master
  git describe: 0.0.3-664-gfab22b9-dirty
    git status:
    --
     M ../../../../third_party/litepcie
     M ../../../../third_party/litesata
     M ../../../../third_party/liteusb
    --

         built: Mar 17 2017 00:23:17
        uptime: 00:04:31
-----------------------------------------------
```

stefanor, 2017-11-02:
Verified the issue is still present in 840e634e, which is 8 commits after Atlys support was added to the nextgen branch. So, it's been there since the beginning.

@enjoy-digital: Any suggestions on approaches to debug it?

mithro, 2017-11-02:
@enjoy-digital 

Is this likely to do with the fact that the CPU or DDR is running at different clock rate than on the Opsis?

You also don't have an Atlys board, right?

enjoy-digital, 2017-11-02:
@mithro: i don't think so since we have an async fifo. I would says it could come from a timing constraint that is not correctly applied on the usb clock domain. What is the difference with the code that was running before on the atlys? (not sure i followed everything)

mithro, 2017-11-02:
@enjoy-digital - As far as we can see, the encoder in the Atlys has never work with the firmware that uses LiteX rather MiSoC -- so a huge amount has changed!

enjoy-digital, 2018-01-21:
I know what it is. It should now be fixed (haven't tested), can someone do a test?

stefanor, 2018-01-25:
So close...

[image]

enjoy-digital, 2018-01-25:
Thanks, just pushed a fix. Want to retry? :)

stefanor, 2018-01-26:
That did it, thanks.

closed 2018-01-26 by stefanor

mithro, 2018-01-26:
YAY!

[image]

enjoy-digital, 2018-01-26:
Note that to speed up place and route ethernet is disabled. If you don't need it, we can keep it like that.

mithro, 2018-01-26:
I don't think the networking is the cause of slow place and route on
Travis. It seems that things just randomly finish in under 30 minutes or
never finish.