timvideos/HDMI2USB-litex-firmware #354: Add 0/1 to EDID monitor name

CarlFK, 2017-09-07:
Currently I see OHW 9" and can't tell which of the 2 inputs that is.

It would be nice if input-0 showed a 0, and 1 1.  

Also OHW is confusing.  something like Opsis or Atlys or HDMI2USB seems better.

9" is also confusing. No idea what won't be confusing.   I am curious what projectors say, as distance to screen determines viewing size.

mithro, 2017-09-08:
https://github.com/timvideos/HDMI2USB-litex-firmware/blob/master/firmware/processor.c#L531-L540

The two inputs are named "HDMI2USB-1" and "HDMI2USB-2" -- this is what they seem to show up with on Mac/Windows. How are you viewing these?

CarlFK, 2017-09-08:
juser@cnt1:~$ hwinfo --monitor --short
monitor:                                                        
                       HDMI2USB-1
                       HDMI2USB-2
                       Acer AL1511

juser@cnt1:~$ hwinfo --monitor 
  Model: "HDMI2USB-1"
  Vendor: OHW 
  Device: eisa 0x5654 "HDMI2USB-1"
  Resolution: 1280x720@60Hz
  Size: 200x112 mm
  Year of Manufacture: 2015

closed 2017-09-08 by CarlFK