timvideos/HDMI2USB-litex-firmware #361: Optimisations to look into

mithro, 2017-09-30:


mithro, 2017-09-30:
UART?
```
INFO:Xst:3231 - The small RAM <Mram_storage> will be implemented on LUTs in order to maximize performance and save block RAM resources. If you want to force its implementation on block, use option/constraint ram_style.
    -----------------------------------------------------------------------
    | ram_type           | Distributed                         |          |
    -----------------------------------------------------------------------
    | Port A                                                              |
    |     aspect ratio   | 16-word x 10-bit                    |          |
    |     clkA           | connected to signal <sys_clk>       | rise     |
    |     weA            | connected to signal <basesoc_uart_tx_fifo_wrport_we> | high     |
    |     addrA          | connected to signal <basesoc_uart_tx_fifo_produce> |          |
    |     diA            | connected to signal <("00",basesoc_interface_dat_w)> |          |
    -----------------------------------------------------------------------
    | Port B                                                              |
    |     aspect ratio   | 16-word x 10-bit                    |          |
    |     addrB          | connected to signal <basesoc_uart_tx_fifo_consume> |          |
    |     doB            | connected to signal <basesoc_uart_tx_fifo_syncfifo_dout> |          |
    -----------------------------------------------------------------------
```

mithro, 2017-09-30:
Something in controllerinjector_bankmachine?
```
INFO:Xst:3218 - HDL ADVISOR - The RAM <Mram_storage_2> will be implemented on LUTs either because you have described an asynchronous read or because of currently unsupported block RAM features. If you have described an asynchronous read, m
aking it synchronous would allow you to take advantage of available block RAM resources, for optimized device usage and improved timings. Please refer to your documentation for coding guidelines.
    -----------------------------------------------------------------------
    | ram_type           | Distributed                         |          |
    -----------------------------------------------------------------------
    | Port A                                                              |
    |     aspect ratio   | 8-word x 24-bit                     |          |
    |     clkA           | connected to signal <sys_clk>       | rise     |
    |     weA            | connected to signal <controllerinjector_bankmachine0_wrport_we> | high     |
    |     addrA          | connected to signal <controllerinjector_bankmachine0_produce> |          |
    |     diA            | connected to signal <("00",_n3595<12:0>,rhs_array_muxed32<8:1>,port_cmd_payload_we)> |          |
    -----------------------------------------------------------------------
    | Port B                                                              |
    |     aspect ratio   | 8-word x 24-bit                     |          |
    |     addrB          | connected to signal <controllerinjector_bankmachine0_consume> |          |
    |     doB            | connected to signal <controllerinjector_bankmachine0_syncfifo0_dout> |          |
    -----------------------------------------------------------------------
INFO:Xst:3218 - HDL ADVISOR - The RAM <Mram_storage_3> will be implemented on LUTs either because you have described an asynchronous read or because of currently unsupported block RAM features. If you have described an asynchronous read, m
aking it synchronous would allow you to take advantage of available block RAM resources, for optimized device usage and improved timings. Please refer to your documentation for coding guidelines.
    -----------------------------------------------------------------------
    | ram_type           | Distributed                         |          |
    -----------------------------------------------------------------------
    | Port A                                                              |
    |     aspect ratio   | 8-word x 24-bit                     |          |
    |     clkA           | connected to signal <sys_clk>       | rise     |
    |     weA            | connected to signal <controllerinjector_bankmachine1_wrport_we> | high     |
    |     addrA          | connected to signal <controllerinjector_bankmachine1_produce> |          |
    |     diA            | connected to signal <("00",_n3595<12:0>,rhs_array_muxed32<8:1>,port_cmd_payload_we)> |          |
    -----------------------------------------------------------------------
    | Port B                                                              |
    |     aspect ratio   | 8-word x 24-bit                     |          |
    |     addrB          | connected to signal <controllerinjector_bankmachine1_consume> |          |
    |     doB            | connected to signal <controllerinjector_bankmachine1_syncfifo1_dout> |          |
    -----------------------------------------------------------------------
```

mithro, 2017-10-05:
@enjoy-digital Do you have any ideas about the error
> XXXX will be implemented on LUTs either because you have described an asynchronous read or because of currently unsupported block RAM features. If you have described an asynchronous read, making it synchronous would allow you to take advantage of available block RAM resources, for optimized device usage and improved timings.
?

enjoy-digital, 2017-10-05:
@mithro : this is not an error, the message is explicit. It seems we are not following the block RAM features (or ISE is not able to recognize it) and we should a closer look at the code to understand what to do. I can have a look but need to know which target you are building.

For the UART, it's a choice that is doing ISE because RAM is small, we can influence that as explain in the message but that's not necessarily better....

mithro, 2017-10-05:
@enjoy-digital Let me rephrase the question;

Is the bank machine in the DDR controller is actually using an asynchronous read or is it a different feature that is causing the issue?

I'm guessing it is because in both cases `Port B` doesn't seem to have a clock attached to it? This seems to come from https://github.com/enjoy-digital/litex/blob/f5a971a8d8b479418faba36c82d0b318be1e6895/litex/gen/genlib/fifo.py#L63 -- specifically the https://github.com/enjoy-digital/litex/blob/f5a971a8d8b479418faba36c82d0b318be1e6895/litex/gen/genlib/fifo.py#L109 ?

What does `fwft` added in https://github.com/m-labs/migen/commit/8baa957539d965b6548983b177e186a855d8ed5e mean?