timvideos/HDMI2USB-litex-firmware #408: surprising stuff when first connecting to Opsis

CarlFK, 2018-02-11:
crelay on/off power cycles the Opsis.

38:49 to 39:00 is 11 seconds to connect - guessing that is the time it takes to boot.

There is less than 1 second (maybe ~0) between connect and "Disabling..."

```
juser@cnt3:~/temp$ crelay 1 on; sleep 5; crelay 1 off; tio /dev/ttyACM0
[tio 07:38:49] tio v1.20
[tio 07:38:49] Press ctrl-t q to quit
[tio 07:39:00] Connected
Disabling input0
Disabling input1
Enabling input1
hdmi_out0: Init I2C...finished.
Disabling output0
Enabling output0
hdmi_out1: Init I2C...finished.
Disabling output1
Enabling output1
H2U 00:00:00>
```
(03:26:30 PM) CarlFK: mithro (IRC): this may not be new.  could be because I have the relay control to power cycle, so I can turn it on, hit up arrow a few times and run the term program in under 10 seconds ....
(03:27:21 PM) CarlFK: mithro (IRC): which if you are power cycling by hand, you have to move around and it may always take longer than 10 seconds to run the term program 
```
juser@cnt3:~/temp$ crelay 1 on; sleep 5; crelay 1 off; sleep 10; tio /dev/ttyACM0
[tio 08:25:00] tio v1.20
[tio 08:25:00] Press ctrl-t q to quit
[tio 08:25:00] Connected
Disabling input0
Disabling input1
Enabling input1
hdmi_out0: Init I2C...finished.
Disabling output0
Enabling output0
hdmi_out1: Init I2C...finished.
Disabling output1
Enabling output1
H2U 00:00:00>
[tio 08:25:05] Disconnected
juser@cnt3:~/temp$ crelay 1 on; sleep 5; crelay 1 off; sleep 12; tio /dev/ttyACM0
[tio 08:25:34] tio v1.20
[tio 08:25:34] Press ctrl-t q to quit
[tio 08:25:34] Connected
[tio 08:44:00] Disconnected
juser@cnt3:~/temp$ 
```

more fun:

```
juser@cnt3:~/temp$ crelay 2 on; sleep 5; crelay 2 off; tio /dev/ttyVIZ0[tio 08:49:02] tio v1.20
[tio 08:49:02] Press ctrl-t q to quit
[tio 08:49:03] Connected
        __   _ __      _  __
                                   / /  (_) /____ | |/_/
                                                              / /__/ / __/ -_)>  <
       /____/_/\__/\__/_/|_|
                                  SoC BIOS / CPU:LM32
                                                     (c) Copyright 2012-2018 Enjoy-Digital
          (c) Copyright 2007-2018 M-Labs Limited
                                                Built Feb 10 2018 01:32:54

                                                                          BIOS CRC passed (2742dbf8)
                    Initializing SDRAM...
                                         Memtest OK
                                                   Booting from serial...
                                                                         Press Q or ESC to abort boot completely.
                                 sL5DdSMmkekro
                                              Timeout
                                                     Booting from flash...
                                                                          Loading 73332 bytes from flash...
                           Executing booted program at 0x40000000
                                                                 HDMI2USB firmware booting...

hardware version info
===============================================
           DNA: 01373178b466ea09
           MAC: Unknown

gateware version info
===============================================
      platform: atlys
        target: hdmi2usb
      revision: d11f5b85e7a962b3b3e55aaee70a693159f59957

firmware version info
===============================================
      platform: atlys
        target: hdmi2usb
    git commit: d11f5b85e7a962b3b3e55aaee70a693159f59957
    git branch: master
  git describe: v0.0.4-144-gd11f5b8
    git status:
    --
    --

         built: Feb 10 2018 01:32:56
        uptime: 00:00:00
-----------------------------------------------
Disabling input0
Disabling input1
Enabling input1
Disabling output0
Enabling output0
Disabling output1
Enabling output1
H2U 00:00:00>
```

mithro, 2018-02-11:
I don't see any corruption here? I just see the fact tio sucks? Try the same with flterm

CarlFK, 2018-02-11:
man tio ...  If  the  device  is not present it will wait for it to  appear and then connect. 

which flterm doesn't do.

```
juser@cnt3:~/temp$ crelay 1 on; sleep 5; crelay 1 off; flterm --port /dev/ttyACM0
[FLTERM] Starting...
Unable to open serial port: No such file or directory
juser@cnt3:~/temp$ crelay 1 on; sleep 5; crelay 1 off; while [ ! -e /dev/ttyACM0 ]; do sleep 0; done; flterm --port /dev/ttyACM0
[FLTERM] Starting...
Unable to open serial port: Permission denied
```

```
juser@cnt3:~/temp$ crelay 1 on; sleep 5; crelay 1 off; while [ ! -e /dev/ttyACM0 ]; do sleep 0; done; sleep .1; flterm --port /dev/ttyACM0
[FLTERM] Starting...
Disabling input0
Disabling input1
Enabling input1
hdmi_out0: Init I2C...finished.
Disabling output0
Enabling output0
hdmi_out1: Init I2C...finished.
Disabling output1
Enabling output1
H2U 00:00:00>
```