timvideos/HDMI2USB-litex-firmware #412: edid broken

CarlFK, 2018-02-28:
v0.0.4-122-gbc7feb5 and xrandr shows connected (and used it)  
looping the Opsis's output to input lets it check its own edid.  seems valid.

v0.0.4-142-gfe99d5b xrandr didn't show a display device connected.  (and thus doesn't use it)

checking loop back: FF FF FF...
Checksum ERROR in EDID block 0

v0.0.4-148-g8810a57 - last build for Opsis, still broken.

checked a few in between, all broken edid, but 4-148 does stream video in and out ok if I hook it to a souce that ignores missing edid. (exatron vga converter)

```
H2U 00:23:49>d edid output0
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
Checksum ERROR in EDID block 0
H2U 00:24:46>d edid output1
00 FF FF FF FF FF FF 00 22 F0 47 29 01 01 01 01 
0D 15 01 03 80 30 1B 78 2E 95 25 A1 59 57 9F 27 
0D 50 54 A1 08 00 81 C0 81 80 95 00 B3 00 D1 C0 
01 01 01 01 01 01 02 3A 80 18 71 38 2D 40 58 2C 
45 00 DD 0C 11 00 00 1E 00 00 00 FD 00 32 4C 18 
5E 11 00 0A 20 20 20 20 20 20 00 00 00 FC 00 48 
50 20 4C 41 32 32 30 36 0A 20 20 20 00 00 00 FF 
00 43 4E 43 31 31 33 30 4B 34 56 0A 20 20 00 DA 
H2U 00:24:53>s
input0:  1280x720 (@74.25 MHz) (capturing)
input1:  0x0 (@0.00 MHz) (capturing)
output0: 1280x720@50.00Hz from input1 (underflows: 0)
output1: 1280x720@50.00Hz from pattern (underflows: 0)
EDID primary mode:   1280x720@50.00Hz
EDID secondary mode: 1280x720@60.00Hz
encoder: 1280x720 @ 0fps from input1 (q: 85)
ddr: 
H2U 00:24:55>version 

hardware version info
===============================================
           DNA: 01332dbbfc5b5aa0
           MAC: d8:80:39:68:0b:42

gateware version info
===============================================
      platform: opsis
        target: hdmi2usb
      revision: fe99d5b8d7d9e530e1ffd7fdc0533d7c2851079b

firmware version info
===============================================
      platform: opsis
        target: hdmi2usb
    git commit: fe99d5b8d7d9e530e1ffd7fdc0533d7c2851079b
    git branch: master
  git describe: v0.0.4-142-gfe99d5b
    git status:
    --
    --

         built: Feb  9 2018 22:37:46
        uptime: 00:25:16
-----------------------------------------------
H2U 00:25:16>
```

mithro, 2018-03-02:
@enjoy-digital - Could you take a quick look at this and see if it was broken from changes required to support @bunnie 's use case?

The only other potential source of problems could have been the edid-decode update. @joeladdison - Maybe you want to have a quick look at this too?

enjoy-digital, 2018-03-02:
@mithro: sorry yes, it was probably broken. https://github.com/enjoy-digital/litevideo/commit/18b88dfee6bf6f4ab55d196747ca00c6c84c2ef2 should fix it.

mithro, 2018-03-19:
As of https://github.com/timvideos/HDMI2USB-litex-firmware/commit/cacbf40f46f1d6ad1bcf3929377a9784ad918a39 @enjoy-digital 's change at https://github.com/enjoy-digital/litevideo/commit/18b88dfee6bf6f4ab55d196747ca00c6c84c2ef2 should be now merged.

@CarlFK can you now confirm EDID is working again?

CarlFK, 2018-03-19:
looks fixed:
desktop box that first showed problems now sees a display device, 
and the opsis likes it's own edid: 
```
H2U 00:00:50>version 

hardware version info
===============================================
           DNA: 01332dbbfc5b5aa0
           MAC: d8:80:39:68:0b:42

gateware version info
===============================================
      platform: opsis
        target: hdmi2usb
      revision: e14edee4c2506080a3d45010fd679d336abbd9a4

firmware version info
===============================================
      platform: opsis
        target: hdmi2usb
    git commit: e14edee4c2506080a3d45010fd679d336abbd9a4
    git branch: master
  git describe: v0.0.4-194-ge14edee
    git status:
    --
    --

         built: Mar 19 2018 03:41:50
        uptime: 00:00:52
-----------------------------------------------
H2U 00:00:52>debug edid output0
00 FF FF FF FF FF FF 00 3D 17 54 56 00 00 00 00 
00 19 01 03 80 14 0B 50 06 00 00 00 00 00 00 00 
00 00 00 00 00 00 01 01 01 01 01 01 01 01 01 01 
01 01 01 01 01 01 01 1D 00 BC 52 D0 1E 20 B8 28 
05 40 C8 70 00 00 00 1E 01 1D 00 72 51 D0 1E 20 
DC 28 14 00 C8 70 00 00 00 1E 00 00 00 FC 00 48 
44 4D 49 32 55 53 42 2D 32 0A 20 20 00 00 00 FD 
00 32 3C 1F 44 08 00 0A 20 20 20 20 20 20 00 51 
H2U 00:01:12>debug edid output1
00 FF FF FF FF FF FF 00 22 F0 47 29 01 01 01 01 
0D 15 01 03 80 30 1B 78 2E 95 25 A1 59 57 9F 27 
0D 50 54 A1 08 00 81 C0 81 80 95 00 B3 00 D1 C0 
01 01 01 01 01 01 02 3A 80 18 71 38 2D 40 58 2C 
45 00 DD 0C 11 00 00 1E 00 00 00 FD 00 32 4C 18 
5E 11 00 0A 20 20 20 20 20 20 00 00 00 FC 00 48 
50 20 4C 41 32 32 30 36 0A 20 20 20 00 00 00 FF 
00 43 4E 43 31 31 33 30 4B 34 56 0A 20 20 00 DA 
H2U 00:01:20>
```

closed 2018-03-19 by CarlFK

mithro, 2018-03-19:
@CarlFK That is great news! Thanks for testing.