timvideos/HDMI2USB-litex-firmware #425: Support flashing to board on nexys_video platform / check flash addresses

FelixHenninger, 2018-04-07:
First of all, thanks for the awesome project! This has been a real joy to tinker with, and I've learnt a huge amount doing so.

I'm currently trying to build and flash the firmware on my Nexys Video board, which builds and runs the firmware and gateware perfectly fine (I'm lagging behind master, FWIW, but the relevant code is the same), and runs nicely in temporary mode. Likewise, flashing the gateware via `make gateware-flash-py` works perfectly, however `make firmware-flash-py` results in the following error:

```
Traceback (most recent call last):
  File "flash.py", line 76, in <module>
    main()
  File "flash.py", line 69, in main
    address_end - address_start, file_size)
AssertionError: File is too big!
-32768 file doesn't fit in 68020 space.
```

This error is thrown by an [assertion in `flash.py`](https://github.com/timvideos/litex-buildenv/blob/master/flash.py#L63), which compares `file_end = address_start + file_size` to `address_end` to guard against overflows, which makes total sense (to me).

However, [further up in `flash.py`](https://github.com/timvideos/litex-buildenv/blob/master/flash.py#L43), where the values are defined for the firmware, looks weird:

```
address_start = platform.gateware_size + make.BIOS_SIZE
address_end = platform.gateware_size
```

Because `make.BIOS_SIZE` on my system is 32768, the result of subtracting `address_end - address_start` is negative that.

There are a few things that seem strange here:
* The `BIOS_SIZE` looks like it should be added to both `address_start` and `address_end` in [line 43](https://github.com/timvideos/litex-buildenv/blob/master/flash.py#L43), and an additional value should be added onto `address_end`. Or, because the firmware comes last on the chip, it might be that line 44 should be `address_end = platform.spiflash_total_size`
* In the assertion in [line 63](https://github.com/timvideos/litex-buildenv/blob/master/flash.py#L63), it seems that the positions of `address_end - address_start` and `file_size` should be replaced. The error message above should read `68020 file doesn't fit in -32768 space.` instead of the other way around.

Ok, I hope this helps, sorry for the wall of text! I'd be happy to help this get fixed, if you can confirm how this is supposed to work, I'll gladly put together a pull request.

mithro, 2018-04-08:
I believe https://github.com/timvideos/HDMI2USB-litex-firmware/pull/426 should fix this.

FelixHenninger, 2018-04-08:
I believe it does, thanks a lot! I've added a minor comment to line 47.