timvideos/HDMI2USB-litex-firmware #430: Atlys - input0 not inputting

CarlFK, 2018-04-17:
image-gateware+bios+firmware.v0.0.4-199-gef22966.atlys.hdmi2usb.lm32.bin

Tried 3 sources: exatron, desktop, pi - all the same:
input0: 0x0
input1: 1280x720 (capturing)

both inputs work on opsis:
input0: 1280x720 (@74.25 MHz) (capturing)
input1: 1280x720 (@74.24 MHz) (capturing)

CarlFK, 2018-04-17:
```
input0 (0): J1
  Type A connector, marked as J1, on side with USB connectors.
  To use J1, make sure:
   * JP4 has a jumper (it connects / disconnects 5V to HDMI pin 18).
   * JP2 (marked only as SDA/SCL - not to be confused with JP6 and JP6)
     has *two* ju
```
[image]

un/plugging and power cyceling does not  cause anything to change:
the same devices work fine on input1

```
H2U 00:57:28>debug input0 on
HDMI Input 0 debug on
H2U 01:40:31>s
input0:  0x0
input1:  1280x720 (disabled)
output0: 1280x720@50.00Hz from input1 (underflows: 0)
output1: 1280x720@50.00Hz from input1 (underflows: 0)
EDID primary mode:   1280x720@50.00Hz
EDID secondary mode: 1280x720@60.00Hz
encoder: 1280x720 @ 0fps from input1 (q: 85)
ddr: 
H2U 01:44:30>i0 on
Enabling input0
H2U 01:45:03>dvisampler0: connected
s
input0:  0x0
input1:  1280x720 (disabled)
output0: 1280x720@50.00Hz from input1 (underflows: 0)
output1: 1280x720@50.00Hz from input1 (underflows: 0)
EDID primary mode:   1280x720@50.00Hz
EDID secondary mode: 1280x720@60.00Hz
encoder: 1280x720 @ 0fps from input1 (q: 85)
ddr: 
H2U 01:45:09>
```

Opsis out - Atlys in0
```
H2U 16:32:05>d edid output0
FF FF FF FF 8F 9F FF FF FF FF FF FF 8F C7 FF FD 
9F 5F FF FF FF FE 83 BF FF 7B BF DF FF FF FF F8 
DF EE DF FF BF FF FF FF F0 3F 9D 86 DF 92 19 9F 
FF FF 3F BF FF FF FF FF BF FF FF 9F FF FF FF FF 
FF FF FF BF 8F EE FF F3 B8 5C CE 66 7F 80 9F DF 
D3 7B 19 08 DF FF E1 BF 9F 5E C2 3F BF FF FF F8 
FF FF 9F 9F FF FF E3 BF FC 27 DF FF FF FE EF D7 
7B BE 1F E7 F7 7E 37 D8 CF F7 FF 1F DF 9F DF E3 
Checksum ERROR in EDID block 0
H2U 16:33:02>d edid output0
43 BB 5F BF FF BF BF FF EF E1 E7 39 6F DF FF FF 
FF FF FA FF FF FF FF DF FF FF FF FC FF FF FF FF 
FD FF FF FF FF 3F FF FF FF FB FF DE EF FF 0F FF 
FF FF FF FF FD DB FF C7 FF FF FE CF F6 7B BF DF 
E3 FF FF FF FE FF B6 7B FF D8 FF FF FF FF FE FF 
FF FF BC 5F FF FF 3B BD FF FF FF FF 3F DF FF FF 
F7 BD DE EF F7 0B 3D DE EF D6 7B FD 3A EC 87 FB 
31 9D CF E7 73 19 9C C2 47 5F 3D DF 9C DF 53 39 
Checksum ERROR in EDID block 0
H2U 16:33:29>d edid output0
FF DF F8 FF FD 5B FF FF FF FF FF FC 67 FF FF FF 
FF FF FF FF FC 3F FF FF FF FF FF FF FF FF 1B FE 
8F FF FF FB 9F FF FF 8F 3F FF BF ED FF FF FF EF 
C3 FF BF 95 FE FF F7 7F FF E0 BF FF FF FF DF FF 
FF FF FC FF FF FF FF FF FF FF FF FF FF FF FF FF 
FF FF FF FF FE 3F FF FF FF FF FF FF FF FF 1F F7 
79 3F FF FF FF FF FF 8F FF FF FF F8 1E FF F7 F1 
C3 BF FF F7 77 7F FF CF FF E1 FB FD FF FF FF FF 
Checksum ERROR in EDID block 0
```

Opsis out - Atlys in1
```
H2U 16:37:19>d edid output0
00 FF FF FF FF FF FF 00 3D 17 54 56 00 00 00 00 
00 19 01 03 80 14 0B 50 06 00 00 00 00 00 00 00 
00 00 00 00 00 00 01 01 01 01 01 01 01 01 01 01 
01 01 01 01 01 01 01 1D 00 BC 52 D0 1E 20 B8 28 
05 40 C8 70 00 00 00 1E 01 1D 00 72 51 D0 1E 20 
DC 28 14 00 C8 70 00 00 00 1E 00 00 00 FC 00 48 
44 4D 49 32 55 53 42 2D 32 0A 20 20 00 00 00 FD 
00 32 3C 1F 44 08 00 0A 20 20 20 20 20 20 00 51 
```

Opsis in-out (for reference, same as above atlys in1)
```
H2U 16:27:42>d edid output0
00 FF FF FF FF FF FF 00 3D 17 54 56 00 00 00 00 
00 19 01 03 80 14 0B 50 06 00 00 00 00 00 00 00 
00 00 00 00 00 00 01 01 01 01 01 01 01 01 01 01 
01 01 01 01 01 01 01 1D 00 BC 52 D0 1E 20 B8 28 
05 40 C8 70 00 00 00 1E 01 1D 00 72 51 D0 1E 20 
DC 28 14 00 C8 70 00 00 00 1E 00 00 00 FC 00 48 
44 4D 49 32 55 53 42 2D 32 0A 20 20 00 00 00 FD 
00 32 3C 1F 44 08 00 0A 20 20 20 20 20 20 00 51 
```
Opsis nothing-out
```
H2U 16:37:26>d edid output0
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
same...
```

mithro, 2018-04-18:
`debug input0 on`

I'm a bit worried about the fact that your inputs don't have a pixel clock value?

CarlFK, 2018-04-18:
Never mind, jumper P5 wasn't jumpering. put it in place, now it works.

 It was on one pin.  the above pic I took to confirm it was in place, but the angle wasn't right.  here is what right looks like, where the shadow covers up the p5 label, not the jumper.  Sorry.

[image]

closed 2018-04-18 by CarlFK

mithro, 2018-04-18:
Can you log a bug about the frequency counter not appearing on the Atlys board?