timvideos/HDMI2USB-litex-firmware #431: usb stream broken

CarlFK, 2018-04-17:
image-gateware+bios+firmware.v0.0.4-197-gdaf33bc.opsis.hdmi2usb.lm32.bin
works:
```juser@cnt3:~/temp$ ./grab_hdmi2usb.sh 
+ gst-launch-1.0 v4l2src device=/dev/video0 num-buffers=4 '!' jpegdec '!' queue '!' videoconvert '!' pngenc '!' multifilesink location=frame-%d.png
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Got EOS from element "pipeline0".
Execution ended after 0:00:00.286815929
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

      platform: opsis
        target: hdmi2usb
    git commit: daf33bc6a62433fde984b7166136585d82dd63de
    git branch: master
  git describe: v0.0.4-197-gdaf33bc
```

image-gateware+bios+firmware.v0.0.4-199-gef22966.opsis.hdmi2usb.lm32.bin
```
juser@cnt3:~/temp$ ./grab_hdmi2usb.sh 
+ gst-launch-1.0 v4l2src device=/dev/video0 num-buffers=4 '!' jpegdec '!' queue '!' videoconvert '!' pngenc '!' multifilesink location=frame-%d.png
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Device '/dev/video0' cannot capture at 1280x720
Additional debug info:
gstv4l2object.c(3458): gst_v4l2_object_set_format_full (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
Call to S_FMT failed for MJPG @ 1280x720: Input/output error
Execution ended after 0:00:44.997937380
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
juser@cnt3:~/temp$ ./grab_hdmi2usb.sh 
```

CarlFK, 2018-09-16:
This seems to be fixed.  not sure anyone knows what broke or fixed it -  any reason not to close the issue?

mithro, 2018-09-17:
Sure?

closed 2018-09-17 by mithro