timvideos/HDMI2USB-litex-firmware #441: Instance 'OSERDESE2_18' of module 'OSERDESE2' requires 27 connections, but only 13 given

mithro, 2018-05-20:
When building the Neso target, there are lots of errors like the following;
```
WARNING: [Synth 8-350] instance 'OSERDESE2_11' of module 'OSERDESE2' requires 27 connections, but only 13 given [/home/travis/build/timvideos/HDMI2USB-litex-firmware/build/neso_base_lm32/gateware/top.v:9792]
WARNING: [Synth 8-350] instance 'OSERDESE2_12' of module 'OSERDESE2' requires 27 connections, but only 13 given [/home/travis/build/timvideos/HDMI2USB-litex-firmware/build/neso_base_lm32/gateware/top.v:9814]
WARNING: [Synth 8-350] instance 'OSERDESE2_13' of module 'OSERDESE2' requires 27 connections, but only 13 given [/home/travis/build/timvideos/HDMI2USB-litex-firmware/build/neso_base_lm32/gateware/top.v:9836]
WARNING: [Synth 8-350] instance 'OSERDESE2_14' of module 'OSERDESE2' requires 27 connections, but only 13 given [/home/travis/build/timvideos/HDMI2USB-litex-firmware/build/neso_base_lm32/gateware/top.v:9858]
WARNING: [Synth 8-350] instance 'OSERDESE2_15' of module 'OSERDESE2' requires 27 connections, but only 13 given [/home/travis/build/timvideos/HDMI2USB-litex-firmware/build/neso_base_lm32/gateware/top.v:9880]
WARNING: [Synth 8-350] instance 'OSERDESE2_16' of module 'OSERDESE2' requires 27 connections, but only 13 given [/home/travis/build/timvideos/HDMI2USB-litex-firmware/build/neso_base_lm32/gateware/top.v:9902]
WARNING: [Synth 8-350] instance 'OSERDESE2_17' of module 'OSERDESE2' requires 27 connections, but only 13 given [/home/travis/build/timvideos/HDMI2USB-litex-firmware/build/neso_base_lm32/gateware/top.v:9924]
WARNING: [Synth 8-350] instance 'OSERDESE2_18' of module 'OSERDESE2' requires 27 connections, but only 13 given [/home/travis/build/timvideos/HDMI2USB-litex-firmware/build/neso_base_lm32/gateware/top.v:9946]
WARNING: [Synth 8-350] instance 'OSERDESE2_19' of module 'OSERDESE2' requires 27 connections, but only 13 given [/home/travis/build/timvideos/HDMI2USB-litex-firmware/build/neso_base_lm32/gateware/top.v:9968]
WARNING: [Synth 8-350] instance 'OSERDESE2_20' of module 'OSERDESE2' requires 27 connections, but only 13 given [/home/travis/build/timvideos/HDMI2USB-litex-firmware/build/neso_base_lm32/gateware/top.v:9990]
```

See the following log - https://travis-ci.org/timvideos/HDMI2USB-litex-firmware/jobs/381124371

mithro, 2018-05-23:
@enjoy-digital Should these extra inputs/outputs be tied to zero / one or something?

enjoy-digital, 2018-05-23:
Vivado/ISE are already doing that for us. We are using this same principle on lots of instances,  so my reasoning is that wether we fix that on all instances or we keep things like this. From a designer perspective i found easier to keep things like that and i'm not sure we really had issue in the past with this.