timvideos/HDMI2USB-litex-firmware #460: Wireshark reports corrupted Frame Check Sequence for Mimas A7

rohitk-singh, 2018-10-10:
Wireshark is reporting incorrect frame check sequence when using Gigabit Ethernet on Mimas A7.
* RGMII interface is being used.
* Autonegotiation succeeds at 1G
* Mimas A7 tries to respond to ARP requests (or maybe broadcasts on its own) as seen in the screenshot below, but wireshark reports FCS as incorrect. Is it a severe issue or should not affect normal functioning?
* Pings are failing. 
* Bootloader also requests for firmware (boot.bin) as shown in the screenshot below, but again the checksum is incorrect. 

@enjoy-digital @mithro Any ideas what could be going wrong here? Maybe issues with RGMII signal delays? Problem is I don't have a spare USB-Serial module to attach a litescope to and probe the RGMII signals. The only USB-Serial interface on MimasA7 is being used for firmware/bootloader console.

[image]

rohitk-singh, 2018-10-10:
I'm planning to debug it via Etherbone target + Litescope without CPU, so that I wouldn't need to use the only UART for console, instead I will use it for Litescope samples extraction. BTW is there any reason Etherbone target for Nexys_video uses **140 MHz** `sys` clock? Should the `sys` clock frequency be >= 125 MHz (or some other reason), or is it just any random value?

rohitk-singh, 2018-10-10:
@enjoy-digital Thanks! I'll take a look. Currently I'm trying to set up a litescope + etherbone design. 

Apologies for not mentioning the code. It is here as part of the pull request: https://github.com/timvideos/HDMI2USB-litex-firmware/pull/447/commits/1dd165e8967e3b6f26087a24435f9e016c32917c

rohitk-singh, 2018-10-10:
@enjoy-digital Did you mean for me to try with different phase values for the rx-clk (here: https://github.com/enjoy-digital/liteeth/blob/master/liteeth/phy/s7rgmii.py#L133) ?

enjoy-digital, 2018-10-10:
You can try the etherbone design first and see what you get when you ping the board.
Then you can try to adjust the phase if it's not working.

If it's not working, i think it will be easier for you to debug with the CPU:
- enable debug here: https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libnet/microudp.c#L13
- disable hardware preamble/crc: https://github.com/enjoy-digital/liteeth/blob/master/liteeth/core/mac/__init__.py#L11 (this will allow you to see exactly with microudp traces what it should transmit and what it receives.)

rohitk-singh, 2018-10-11:
@enjoy-digital Thanks for the great tips! The interesting thing is that etherbone design seems to be responding to pings. Here is the VCD dump for Ping request and response. 

[dump.zip](https://github.com/timvideos/HDMI2USB-litex-firmware/files/2469729/dump.zip)
[image]
[image]

Maybe something is wrong with the TCP/IP stack. I'll try with enabling debug on uIP.

rohitk-singh, 2018-10-11:
@enjoy-digital Sure, I'll try with enabling debug in microudp. I tried with serialboot till now. I'll try with tftp as per your recommendation. Pings were failing but I could see ARP and TFTP firmware request packets from boards in wireshark. 

Thank you so much for all the help and great tips! It's been real life-saver! :)

rohitk-singh, 2018-10-11:
@enjoy-digital It's working. Very weird. Last night I tried everything to get it working but it didn't. After following your structured steps for debugging it is now working without any issues and modifications! Thank you so much for your help! :)

**Current Status**
* TFTP boot is working
* Pings working
* Telnet console is working 

One more question: How do I use the etherbone interface of HDMI2USB firmware? I tried to use Litex server, which failed to read/write memory. Upon looking into the code, I found 2 interesting things, 1. Etherbone on HDMI2USB is implemented purely on software (which I was not expecting at all, ingenious idea!), 2. It uses TCP protocol, but `Litex Server` uses UDP protocol, so it wasn't working. 

So, how do we actually use it? Is there any TCP client for Etherbone? cc: @mithro

enjoy-digital, 2018-10-11:
I think it should simply work by doing in your script: wb = RemoteClient("192.168.1.50") :)
(just maybe also verify the port)

rohitk-singh, 2018-10-11:
OMG, LOL. Indeed it works. Thanks @enjoy-digital ! :) 

Closing this issue.

closed 2018-10-11 by rohitk-singh