timvideos/HDMI2USB-litex-firmware #464: Support chubby75 board

danielkucera, 2019-01-18:
Hi,

would it be possible to support https://github.com/q3k/chubby75 ?
I can try if you guide me a little bit. (I will also try if you won't)

mithro, 2019-01-18:
We would love to support the board! I order a bunch myself but have not had any time to look at them.

I would recommend jumping on IRC and asking questions in the #timvideos channel.

danielkucera, 2019-06-05:
my first try:
https://github.com/danielkucera/HDMI2USB-litex-firmware/commit/00ed664c7bf683f86777fcd526bf08b3a90b4694
it builds but I don't know how to continue. I can write `top.bin` to SPI flash but there is no uart defined. How do I do get some "shell" ?

mithro, 2019-06-05:
@danielkucera If you know the net configuration works, you could try tftp or telnet?

mithro, 2019-06-05:
I created a WIP pull request at https://github.com/timvideos/HDMI2USB-litex-firmware/pull/473 so we can see the progress.