timvideos/HDMI2USB-litex-firmware #465: encoder broken: JPEG file: starts with 0x15 0x0c

CarlFK, 2019-01-19:
last good version: v0.0.4-399-g33b53ba  (works)
next version is v0.0.4-404-g57fb821 (broken)

juser@dc10b:~$ gst-launch-1.0 v4l2src device=/dev/video2 ! jpegdec ! fakesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstJpegDec:jpegdec0: Failed to decode JPEG image
Additional debug info:
gstjpegdec.c(1394): gst_jpeg_dec_handle_frame (): /GstPipeline:pipeline0/GstJpegDec:jpegdec0:
Decode error #53: Not a JPEG file: starts with 0x15 0x0c
Execution ended after 0:00:00.434340910
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

CarlFK, 2019-02-20:
v0.0.4-455-gde59df1

1, different error
2. 2 video devices  - goes back to v0.0.4-399-g33b53ba, maybe before.

```
(venv) juser@gator:~/crelay/src$ ls /dev/video*
/dev/video0  /dev/video1
(venv) juser@gator:~/crelay/src$ gst-launch-1.0 v4l2src device=/dev/video0 ! jpegdec ! fakesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstJpegDec:jpegdec0: Failed to decode JPEG image
Additional debug info:
gstjpegdec.c(1394): gst_jpeg_dec_handle_frame (): /GstPipeline:pipeline0/GstJpegDec:jpegdec0:
Decode error #55: Not a JPEG file: starts with 0xff 0xde
Execution ended after 0:00:00.494827912
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
(venv) juser@gator:~/crelay/src$ gst-launch-1.0 v4l2src device=/dev/video1 ! jpegdec ! fakesink
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Device '/dev/video1' is not a capture device.
Additional debug info:
v4l2_calls.c(631): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
Capabilities: 0x4a00000
Setting pipeline to NULL ...
Freeing pipeline ...
```
```
(venv) juser@gator:~/crelay/src$ sudo dmesg 
[53373.686740] usb 1-5: new high-speed USB device number 38 using xhci_hcd
[53373.834856] usb 1-5: New USB device found, idVendor=2a19, idProduct=5440, bcdDevice= 0.00
[53373.834863] usb 1-5: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[53380.682018] usb 1-5: USB disconnect, device number 38
[53381.026546] usb 1-5: new high-speed USB device number 39 using xhci_hcd
[53381.174676] usb 1-5: New USB device found, idVendor=2a19, idProduct=5440, bcdDevice= 0.00
[53381.174682] usb 1-5: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[53381.490638] usb 1-5: USB disconnect, device number 39
[53383.086481] usb 1-5: new high-speed USB device number 40 using xhci_hcd
[53383.235109] usb 1-5: config 1 interface 2 altsetting 0 endpoint 0x81 has an invalid bInterval 64, changing to 10
[53383.235745] usb 1-5: New USB device found, idVendor=2a19, idProduct=5442, bcdDevice= 0.02
[53383.235751] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[53383.235756] usb 1-5: Product: HDMI2USB.tv - Numato Opsis Board
[53383.235760] usb 1-5: Manufacturer: TimVideos.us
[53383.235764] usb 1-5: SerialNumber: ffffd8803967d4f5
[53383.237882] uvcvideo: Found UVC 1.00 device HDMI2USB.tv - Numato Opsis Board (2a19:5442)
[53383.238061] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
[53383.238478] uvcvideo 1-5:1.0: Entity type for entity Extension 3 was not initialized!
[53383.238486] uvcvideo 1-5:1.0: Entity type for entity Processing 2 was not initialized!
[53383.238491] uvcvideo 1-5:1.0: Entity type for entity Camera 1 was not initialized!
[53383.239003] cdc_acm 1-5:1.2: ttyACM0: USB ACM device
```

stefanor, 2019-07-21:
Yeah, just hit the same thing:

> There are only 'skip'ped commits left to test.
> The first bad commit could be any of:
> 2f9efff8870bc94da7e444e5871262b50ceaaf5e
> 4f23e4e8197652adc12abbc9208118bd4cd9733e
> 5760a1650569f78faadddf5c1c055949bb17815b
> We cannot bisect more!

Those include litex submodule updates. So a lot more in there to test...

mithro, 2019-07-22:
Can you confirm that `v0.0.4-399-g33b53ba` is the last version that works correctly?

CarlFK, 2019-07-22:
v0.0.4-399-g33b53ba errors:

pi@oppi:~/litex-buildenv-jenkins/hdmi2usb/tests $ gst-launch-1.0 v4l2src num-buffers=3 device=/dev/hdmi2usb/by-num/opsis0/video ! jpegdec ! fakesink
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Device '/dev/hdmi2usb/by-num/opsis0/video' is not a capture device.
Additional debug info:
v4l2_calls.c(633): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
Capabilities: 0x4a00000

mithro, 2019-07-22:
So, when is a revision that actually does work?

CarlFK, 2019-07-22:
this suggests it works: 
(testing it again now...)
https://veyepar.nextdayvideo.com/static/temp/opsis-v0.0.4-487-g5940dcb.png

for sure this is solid - used at quite a few shows:
v0.0.4-340-g3fb4e7c

stefanor, 2022-05-27:
I think this was fixed? [v0.0.4-721-gc27ca25](https://github.com/timvideos/HDMI2USB-firmware-prebuilt/tree/master/archive/master/v0.0.4-721-gc27ca25) seems to work

closed 2022-05-27 by stefanor