timvideos/HDMI2USB-litex-firmware #466: revs.txt out of date

CarlFK, 2019-02-10:
v0.0.4-453-gfd68613 exists (buggy) 

but the last entries in
https://code.timvideos.us/HDMI2USB-firmware-prebuilt/revs.txt
```
v0.0.4-364-g0eab730
v0.0.4-368-gd2a01e9
v0.0.4-379-gb29905d
v0.0.4-395-g585091a
v0.0.4-399-g33b53ba
v0.0.4-404-g57fb821
v0.0.4-418-ga33050b
v0.0.4-436-g4383c27
```

I think this is the code that does it?

https://github.com/timvideos/HDMI2USB-litex-firmware/blob/master/.travis/generate-prebuilt-list.py

mithro, 2019-02-11:
Yes, that `generate-prebuilt-list.py` script is what updates that. It gets run as part of the travis build. See the https://github.com/timvideos/HDMI2USB-litex-firmware/blob/master/.travis/update-prebuilt-list.sh and https://github.com/timvideos/HDMI2USB-litex-firmware/blob/fd686136f9bb7e9ebf902049331446854974fc03/.travis.yml#L35-L36

You can see if it is working by looking at the logs from the travis builds here -> https://travis-ci.org/timvideos/HDMI2USB-litex-firmware

Random sample says the following is happening;
```

after_success
240.77s$ $PWD/.travis/update-prebuilt-list.sh
Updating prebuilt binaries
=============================================
Running with PLATFORMS='opsis'
Running with TARGETS='video'
Running with CPUS='lm32'
svn: E175002: Unexpected HTTP status 504 'Gateway Timeout' on '/timvideos/HDMI2USB-firmware-prebuilt/!svn/vcc/default'
svn: E175002: Additional errors:
svn: E175002: REPORT request on '/timvideos/HDMI2USB-firmware-prebuilt/!svn/vcc/default' failed: 504 Gateway Timeout
Looking at cpu 'lm32' in target 'video' for platform 'opsis' from 'timvideos/HDMI2USB-firmware-prebuilt'
---------------------------------------------------------------------------
Found 152 revisions, 10 latest:
  * v0.0.4-453-gfd68613
  * v0.0.4-451-g72a58c2
  * v0.0.4-449-gdb412c4
  * v0.0.4-444-gf271c2a
  * v0.0.4-442-gbe21a99
  * v0.0.4-436-g4383c27
  * v0.0.4-418-ga33050b
  * v0.0.4-404-g57fb821
  * v0.0.4-399-g33b53ba
  * v0.0.4-395-g585091a
Getting other channels..
Channel unstable   is at rev v0.0.4-453-gfd68613
Using /tmp/tmpbjoo1i2w
Download github pages repo...
Traceback (most recent call last):
  File ".travis/generate-prebuilt-list.py", line 186, in <module>
    "svn checkout {} -q {}".format(svn_args, pre_base_url).split())
  File "/usr/lib/python3.4/subprocess.py", line 561, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['svn', 'checkout', '--non-interactive', '--username', 'timvideos-robot', '--password', '[secure]', '-q', 'https://github.com/timvideos/HDMI2USB-firmware-prebuilt/branches/gh-pages']' returned non-zero exit status 1
Update succeed!
Push finished!
```

CarlFK, 2019-02-11:
I'm guessing this is a gh problem - I'll search around and see what I I can find.
```
$ svn checkout --username timvideos-robot https://github.com/timvideos/HDMI2USB-firmware-prebuilt/branches/gh-pages
svn: E175002: Unexpected HTTP status 504 'Gateway Timeout' on '/timvideos/HDMI2USB-firmware-prebuilt/!svn/vcc/default'
```

CarlFK, 2019-02-11:
this works:
svn checkout https://github.com/carlfk/veyepar

$ time svn checkout https://github.com/timvideos/HDMI2USB-firmware-prebuilt/branches/gh-pages
svn: E175002: Unexpected HTTP status 504 'Gateway Timeout' on '/timvideos/HDMI2USB-firmware-prebuilt/!svn/vcc/default'
132.310 seconds 

I'm guessing the size of the prebuilt repo is the problem.    the svn interface may start by cloning the repo, which I think took 20 min.  (or maybe that was checking out linux)

Can you point this at a different repo?