timvideos/HDMI2USB-litex-firmware #470: left row of pixels duplicated on hdmi out

CarlFK, 2019-06-02:
looks like all the pixels are shifted to the right, 
there is a extra copy of the left side, and the right side is lostjj

the error is not present in the encoder output,  so I guess it is correct in the frame buffer, but the  buffer to hdmi is off.

I see it in both Atlys and Opsis, both of these revs (I didn't go back further) 

  git describe: v0.0.4-485-g5fbb99d
and
  git describe: v0.0.3-1045-gccb30b9-dirty

The pictures below: 

bw diagonal lines - dupe pixels are easy to see.  I can see them on my LCD monitor too.

ellipse: 

the upper 1/2 is the original, the lower half what is captured from a 2nd hdmi2usb device.

[image]

[image]

mithro, 2019-06-02:
Can you be more specific about "left row"? Do you mean `x=0`?

CarlFK, 2019-06-02:
Yes. x=0.