timvideos/HDMI2USB-litex-firmware #472: Very confusing documentation

danielkucera, 2019-06-05:
- main README.md is all about litex-buildenv and all links lead to litex wiki which I did not notice first few times and confused me a lot.
- scripts/bootstrap.sh contains 
```
if [ -z "$GITHUB_REPO" ]; then
  GITHUB_REPO=litex-buildenv
fi
```
so it installs litex-buildenv too.

I suggest removing the main README.md and putting in just a quickstart:
```
export GITHUB_REPO=HDMI2USB-litex-firmware
curl -fsS https://raw.githubusercontent.com/timvideos/$GITHUB_REPO/master/scripts/bootstrap.sh | bash
cd $GITHUB_REPO/

export PLATFORM=opsis
source scripts/enter-env.sh
make gateware
```

mithro, 2019-06-05:
Hi @danielkucera,

Sorry about the confusion, things are currently in a state of flux. The environment was originally developed for the HDMI2USB project but has become useful for other things like FuPy, Linux, etc and hence the LiteX-BuildEnv was created.

Currently the HDMI2USB-litex-firmware and litex-buildenv contain the same code but the hope is this repository will eventually contain only HDMI2USB specific stuff and litex-buildenv will contain all the code shared between all the projects.

Does that make sense?

Tim '@mithro' Ansell

danielkucera, 2019-06-05:
Sure, I figured it out at the end. Just my 50 cents to this topic.